jdolinay/avr_debug #2: Port to other boards.

martinmarcos, 2016-11-07:
I think this project is great! I would so much need this for my mega2560 board.

I was thinking that it would be great if it had a porting layer so that support for other boards could be added easily. Do you think that is possible?

Porting layer or not, how would go about porting the stub to the mega2560?

Cheers.

jdolinay, 2016-11-08:
Thank you! I am thinking about port for mega2560 too but still haven't found the time... 

If you can spend about $50 it is possible to debug Arduino mega via the JTAG interface using AVR Dragon or similar tool. There is/was a nice article about this here (http://automation.binarysage.net/?p=1515), but the page seems to be down now. 

If you want to port this software debugger, I think it would be best to start by porting the UART communication code - uart_init, getDebugChar etc. Plus the ISR. Then the code for emulating software interrupt in gdb_enable_swinterrupt and gdb_disable_swinterrupt and code which works with the registers and memory such as gdb_read_registers etc. I am not sure how much the registers differ on mega2560, maybe not at all. 

When I get into this I will try to create a porting layer and let you know. If you get this working first, please let me know too :)

jdolinay, 2017-01-30:
Hi, the avr_debug stub now supports Arduino Mega with mega2560 and mega1280. 

As for porting layer, I realized that the "stub" should be a single file for each architecture. The mega2560 is the same architecture as mega328 - 8-bit AVR. So I decided to make the changes in the same file just using conditional compilation. There were only few changes anyway.

closed 2019-02-21 by jdolinay