jdolinay/avr_debug #3: gdb 'next' a few times, then aborts (quits)

JungHong, 2018-05-10:
The following is about using Uno, eclipse, and the current version of avr_debug, following your 69 page instruction "Arduino Debugger"  (dated Jan 18, 2018).  I am on a Windows 10.

I had to revert until eclipse luna SR2 of June 2015, else, with later versions of eclipse, I could not get past page 26 of your instructions:  Specifically, in the properties for the test project, "AVR ..." did not show until I used luna SR2.

My Arduino Uno shows up on COM5. Arduino 1.8.5 builds, up loads programs, and executes ok.

In the eclipse environment per your "Arduino Debugger", the example from page 37 compiles, up loads, and I can do a few "step over" before it quits, halts. Similarly, the example from page 40 compiles, up loads, and I can do a few "step over" before it too halts.  The number of times I can repeat varies, a few to a dozen, maybe, but I was not able to do as many as 20.

In command line mode, I used the avr-gdb.exe from Atmel Studio 7.0 and I am able to connect with the UNO with the above example loaded.  I am able to do several "n" commands,  but after a number of times, I get "Remote connection closed", then the message, "The program is not being run".

What could it be?  If the Gnu Gdb RSP protocol (your avr_debug in the UNO) is stable, then it must be the communications link.

Is there some way to get meaningful, verbose messages as to what is happening?

Any assistance would be appreciated, since being able to debug programs running on the Uno via gnu gdb would be quit useful to me.

Thank you.

jdolinay, 2018-05-11:
This really looks like communication problem. Are you using the TCP-Serial proxy or direct serial connection? 
On Window 10 direct serial connection should work, so preferably use this and not the TCP-Serial proxy. But if you get this error while using direct serial connection, try the TCP-serial proxy and let me know what happens.

One way to diagnose this is to use some serial port monitoring software to see the communication byte by byte. I use Device Monitoring Studio (https://www.hhdsoftware.com/). When the direct serial connection failed on Window 7 it was because there was End-Of-File (EOF) character sent from the serial port to GDB and GDB closed the connection. I think the EOF came from the Windows USB-Serial driver for Arduino because the avr_debug never sends it. For this reason I used the TCP-Serial proxy at that time. But on Win10 it works directly over serial line for me. But this may depend on the your Uno board version (clone) and the USB-serial chip it uses and consequently the driver which Windows uses for your board.

As for eclipse, I am running it in new Eclipse Oxygen 64 bit on Win10. If you did not see the AVR properties for your project it could be because the AVR eclipse plugin was not active for your project. I had similar problem because I created wrong type of project. In the new Eclipse they changed the New project commands. To create project for Arduino (AVR), use File > New > "C++ Project" or "C project" but do not use the "C/C++ Project". The select AVR Cross Target application > Empty project. Then you should see the AVR category in project properties. 
If you select New > "C/C++ project" which gets you to the Managed Build... selection, the project does not contain AVR properties. I guess it is because the "managed build" projects are not supported by the AVR eclipse plugin.

JungHong, 2018-05-11:
Thanks for your reply.  I will try Exlipse Oxygen 64 per your suggestions: use File > New > "C++ Project" .

jdolinay, 2018-05-14:
Thanks for the report and also for the info about COMbyTCP.exe - I didn't know this program. I think it will work in eclipse if it works in command line.

closed 2018-05-14 by jdolinay