jdolinay/avr_debug #5: millis() allways returns 0 under debug active

Kabron287, 2019-12-21:
millis() allways returns 0 under debug active

jdolinay, 2019-12-22:
Thanks for the report. Unfortunately this is most likely a side effect of the way the debugger is implemented. The millis() use timer interrupt to increment the count of milliseconds but the interrupt is not executed if there is some breakpoint inserted in the code. This is because the external interrupt which is used to implement the debugger has higher priority. I think the millis() should work if you use the "flash breakpoints" mode instead of the default "ram breakpoints". But I have to investigate and document the behavior of the debugger with millis() yet.

jdolinay, 2021-03-02:
I updated the doc with information about the millis and micros behavior. It really does not return meaningful values in the RAM breakpoints mode because of the implementation as mentioned above. The only case it works somehow is if you remove all breakpoints and let the program run at full speed (with the Continue button). It may be possible to debug somehow this way with using the hard-coded breakpoint function inside the condition which depends on the millis or micros value, but the best option is to switch to flash breakpoints mode where it works quite well.

closed 2021-03-02 by jdolinay