jdolinay/avr_debug #6: VS2019 Not openning debugger

AbbottHMG, 2020-01-11:
I've been working with Visual Micro about this issue for a while. They suggested I check with you. When I upload my large project (even a simple project) the debugger does not start. It does start on their system with my code). I've even re-installed the latest versions of vs2019 and Visual Micro.
Are there any special configuration setting (vMicro or VS2019) that I should be looking at.
I've attached my vs project and the screen shot after the upload.
[image]
[AvrTest.zip](https://github.com/jdolinay/avr_debug/files/4049247/AvrTest.zip)
I've removed the Debug and .vs folders to reduce file size.
Should I be looking at another environment?
Thanks
Abbott

AbbottHMG, 2020-01-12:
I made the test simpler (by removing everything but avr.
When  i add the breakpoint in Loop, the setup breakpoint is never hit.

The same exception now occurs in the loop breakpoint.

I can F10 to next, when it gets to delay, F10 steps into instead of over.
[MaplePumpDebug2.zip](https://github.com/jdolinay/avr_debug/files/4050578/MaplePumpDebug2.zip)

I've attached the solution.

Abbott

jdolinay, 2020-01-13:
Hello. I didn't know this debugger can be used in VS. I tested it only in Eclipse and Visual Studio Code.

One thing to try first for the breakpoint() not hit in setup:
- try to add sei(); after the call to debug_init() but before the breakpoint(); call. This will enable interrupts. The debugger needs interrupts enabled to work but I guess the Arduino libraries enable them only after your code returns from setup(). Just guessing... Or, if possible, place the breakpoint() into loop() rather than into setup() if it seems that enabling the interrupts "too soon" disrupts the working of other Arduino code. 
 
Also, many problems can be caused by compiler optimizations - try to disable them. I don't know how this is handled in the VS with Visual Micro but if it is similar to VS Code (or in general, if it uses the Arduino IDE build environment as VS Code does) then you can look at this article: https://www.codeproject.com/Articles/5160447/Creating-and-Debugging-Arduino-Programs-in-Visua-2 for instructions how to turn off the optimizations. 

In VS Code if I debug code with optimizations enabled it shows exception on a breakpoint in the Loop(); Code with optimizations off doesn't show this exception.

The F10 doing step into instead of over could also be because of the optimizations. Or it is because the compiler generates the code differently than you wrote it - this sometimes happens even with optimizations off. I have some example in the avr_debug.pdf document in the doc subfolder in the avr_debug folder - see the "Problem: The program is not jumping at some line or is jumping 
somewhere I don’t want it to…".

Another points to consider:
- I saw some situations when the program didn't stop on the breakpoint() function,  but I wasn't able to figure out why. Maybe the program is "running" but when you click the pause/suspend button it may be stopped on the breakpoint line? 

- the IDE actually connects to a running program, so if you insert breakpoint from the IDE (the bullet; not the breakpoint() function call) it will never be hit in setup() because the setup runs right after the program is uploaded/reset and by the time the debugger connects and sets the breakpoint, the program is already past the setup() (and also through the loop many times also).

By the way, the Visual Micro seems to contain their own debugger. Did you try this debugger?

Jan

AbbottHMG, 2020-01-13:
Thanks. I’ll follow thru with your suggestions. My gut feel right now is my use of enums … possibly.

 

Vm promotes your debugger. My program is quite large, so I’m not trying to get thru a little at a time.

 

Abbott

AbbottHMG, 2020-03-01:
Thanks for everyone's input and help
Abbott

closed 2020-03-01 by AbbottHMG