jdolinay/avr_debug #7: Exception has occurred. Unknow stopping event

wanggaoteng, 2020-01-15:
It's an excellent project. Following the instructions in the "doc/avr_debug.pdf", I debugging arduino uno successfully in vscode. But everytime the "debug arrow" returns to the loop() function from the main() function, an unexpected message appears, it says "Exception has occurred. Unknow stopping event".

(https://user-images.githubusercontent.com/29191854/72438764-7e562c00-37e0-11ea-80c9-554bab11c610.jpg)

jdolinay, 2020-01-15:
Thank you! I saw this error too and found out it goes away if the compiler optimizations are turned off. In the default Arduino configuration with optimizations enabled, this error occurs, without them there is no error. 

Turning off the optimizations is described in the doc in chapter "Step 3: Turn off compiler optimizations", page 72. It's easy to place the file into wrong location so please make sure it really works - the program size should increase considerably when optimizations are off. You can test it by commenting out the lines in the platform.local.txt file - just insert # at the beginning of the lines - the program should be smaller in this case because optimizations are on. If you un-comment the lines the program should be much larger. 

As for the reason, I found out that when stepping from main() to loop() function there is actually no step because the loop is part of the main because of optimizations. So what happens (I guess) is that VS Code asks the GDB debugger  (avr-gdb.exe) to perform the step to loop. But GDB sees there is no step needed - it is already in the loop() so it just sends some signal indicating that the step was not needed(?). But VS Code does not recognize this signal and shows the error "unknown stopping event". 

This is my guess what happens :) I think the GDB extension for VS code would need to be modified to handle this "unknown" error code. There is nothing the avr-debug library in our Arduino can do about this. This is communication problem between the VS Code debug extension and the avr GDB. 

In worst case you can just ignore this error message and the debugging should work normally.

wanggaoteng, 2020-01-16:
Hi, Jan,
Thank you for giving me so detailed answer.
I copyed the "platform.local.txt" file to the path where arduino ide installed (in my computer, the path locate in D:\arduino\hardware\arduino\avr) yesterday, but the optimizations aren't off, even I change the "-Os" to "-O0" in "platform.txt" of the same path.
Today, I recall that there is another path: C:\Users\wgt04\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2
I copy the "platform.local.txt" to this path, and the optimizations aren off as expected, the program size changes from 4774 byte to 7884 byte.

So..., the "Exception has occurred. Unknow stopping event" message disappeared when debugging.
 
Best regards.

jdolinay, 2020-01-16:
Hi, thanks for letting me know the result. So I close this issue :) 
Maybe your VS Code is using some older version of Arduino which is located in the AppData\Local... path - you could check the preferences of the VS Code Arduino extension. In my case it worked with the "normal" arduino path in Program Files, but I haven't updated to Arduino 1.8.10 yet.

Best regards
Jan

closed 2020-01-16 by jdolinay