jdolinay/avr_debug #8: A way to use software serial in debugging

wanggaoteng, 2020-01-16:
Hi, Jan,
There is a problem when using the software serial in debugging: 
I have a little usb-serial board (base on CH340) in hand, there is my connect:
arduino uno                          CH340 board
D2                                          TXD
D3                                          RXD
GND                                       GND
Then I connect both of them's usb ports to my computer using two usb cables, and find that the serial port of arduino is COM9, the serial port of CH340 board is COM7.
The code:
`
#define _DEBUG 1

#ifdef _DEBUG
    #include <SoftwareSerial.h>
    #include <avr8-stub.h>
    #include <app_api.h>
    #define USE_SERIAL SWSerial 
    SoftwareSerial USE_SERIAL(2,3);
#else 
    #define USE_SERIAL Serial
#endif 

const int NUMBER_OF_FIELDS=3;
int fieldIndex=0;
int values[NUMBER_OF_FIELDS]={0};

void setup()
{
    #ifdef _DEBUG
    debug_init();
    #endif 
    USE_SERIAL.begin(9600);
}

void loop()
{
    USE_SERIAL.println(fieldIndex);//just for test
    if(USE_SERIAL.available())
    {
        char ch=USE_SERIAL.read();
        if(ch>='0'&&ch<='9')
        {
            if(fieldIndex<NUMBER_OF_FIELDS)
            {
                values[fieldIndex]=(values[fieldIndex]*10)+(ch-'0');
            }
        }
        else if(ch==',')
        {
            fieldIndex++;
        }
        else 
        {
            for(int i=0;i<min(NUMBER_OF_FIELDS,fieldIndex);i++)
            {
                USE_SERIAL.println(values[i]);
                values[i]=0;
            }
            fieldIndex=0;
        }
    }
}
`
Step1: Select COM9 in vscode, then upload the code above to arduino.
Step2: Click debug button, then the vscode enter the debugging interface.
Step3: Close COM9, select and open COM7 in vscode, and select the baud rate 9600.
Step4: Click the step over button, then the software serial port output appears (you can see the 0 in the OUTPUT tab). So the output of COM7 is OK in debugging.

My problem is: Is there anyway to send data to COM7 in debugging? I press "F1" key and select "Arduino: Send Text to Serial Port", and enter "12,3,1", and press the "enter" key, but it doesn't work. It seems that USE_SERIAL.available() always return 0, so the "debug arrow" can't enter the "if(USE_SERIAL.available())" statement.

Best regards.

jdolinay, 2020-01-17:
Hi, 
I think Software serial internally uses interrupt for receiving data and unfortunately the interrupts are blocked by the debugger if there is one or more breakpoints set. So the Software serial cannot receive data.  This is due to the implementation of the debugger. I can suggest 2 options in your situation to bypass this limitation: 
1. If you remove all breakpoints (including the automatically inserted breakpoint in main which VS code inserts - this requires executing command -exec clear main in the Debug Console, see the doc, chapter Breakpoint in main). So if you remove all breakpoints and let the program run, it should be able to receive data. The problem is that you can only stop it by the pause (suspend) button so it will stop at random moment. But maybe this can still be useful to inspect the data. Actually, you can also insert the breakpoint() function in your code at the line in code where it should stop and it should stop at this line always. Perhaps inside some condition which is true when complete data is received. 

2. If you use the "flash breakpoints" configuration (which requires replacing the bootloader in your arduino), it should work with breakpoins - you can place breakpoint and let the program run. It should then stop on the breakpoint. 

Anyway, I am not sure if it will work :( The software serial may need quite precise timing of its operations to be able to sample the input pin which may be disrupted by the debugger. I think it could work as described above but maybe this is something for which the debugger is not useful.

Best regards

wanggaoteng, 2020-01-17:
Hi, Jan,
Thank you very much. I test 1, but it doesn't work.
I realized that it may be the limitation of gdbstub to receiving data, so I can only close this issue.
Best regards.

closed 2020-01-17 by wanggaoteng