jdolinay/avr_debug #11: Support for other development tools

msquirogac, 2020-05-26:
I've tested avr-debugger with the [PlatformIO](https://docs.platformio.org/en/latest/what-is-platformio.html) development tools, and works fine for both coding and debugging throught your library. Maybe you should consider updating the manual document and examples and include said tool, and if you need it, I got a repository with a ready to use [example](https://github.com/msquirogac/platformio-avr-debugger-example).

jdolinay, 2020-05-31:
I updated the manual, there is now a chapter about using the tool with PlatformIO. Please take a look and let me know what you think. 
Thanks for the link to your article, I found it just now so I'll have a look...

msquirogac, 2020-05-31:
That's great!
Now, we been [discussing](https://community.platformio.org/t/another-tool-for-debugging-avr-microcontrollers/13776/4?u=msquirogac) about the compatibility outside Arduino. Right now, the `lib_compat_mode` parameter is needed when building without the Arduino framework and we been thinking on how to make it work out of the box without requiring said parameter.
The problem comes from the fact that the library was registered as an Arduino one and doesn't have a library.json but a [library.properties](https://github.com/jdolinay/avr_debug/blob/master/arduino/library/avr-debugger/library.properties) file.
To fix this a [library.json](https://docs.platformio.org/en/latest/librarymanager/config.html#library-json) must be placed somewhere in the repository, then we can ask the PlatformIO dev team to use the new file.

jdolinay, 2020-06-01:
OK. Would it work if I place library.json file into the avr_debug\arduino\library\avr-debugger?  
If you have the library.json file I will place it there. Or I can try to prepare it myself. 
As for the code itself, it does not depend on Arduino framework, it works with "plain C" program also. This is actually what I recommend in the manual as the first step when setting it up in eclipse - to start with simple blink without Arduino framework.

In the discussion you also mention Arduino Mega. It should work with both 1280 and 2560 but note that only in the RAM breakpoint mode. I didn't get to modifying the code and bootloader for Mega :(

msquirogac, 2020-06-02:
Hi, I got some questions about the library
1) How many breakpoints are supported?
2) Does it support watchpoints?

Asking because right now I'm tweaking the gdb configuration and these parameters
~~~
set remote hardware-breakpoint-limit 4
set remote hardware-watchpoint-limit 4
~~~

Also, I think we need support for the gdb packet [qRcmd](https://sourceware.org/gdb/current/onlinedocs/gdb/General-Query-Packets.html) which is used by the [monitor](https://sourceware.org/gdb/current/onlinedocs/gdb/Server.html#Monitor-Commands-for-gdbserver-1) command. VSCode in debug mode has a **restart** and **stop** buttons which 
send the `monitor reset` and `monitor halt` to GDB.
The former should reset the program execution and the later detach the debugger.

jdolinay, 2020-06-09:
Hi, sorry by mistake I send reply to the github notification email instead posting it here :( Just found out now when about to post update about the qRcm packet.
First here is the old reply, then I\ll post the update in next message:

Hi, the number of breakpoints can be set in avr8-stub.h, in the AVR8_MAX_BREAKS constant. Default is 8 for RAM mode and 4 for flash breakpoint mode. The higher the number the more RAM is used by the library and it will probably slow down the program a bit more in the RAM mode. Other than that there is no actual limit.

Watchpoints are not supported. Thinking about it it would be possible in the RAM mode to check some address for change after every instruction just as it now checks whether some breakpoint is reached, but in flash bp mode I don't see a way to implement this. I guess it uses some on-chip debug hardware on those MCUs which support this.

I will look at the qRcmd packet. It should be quite easy to support it, but not sure how much this is useful. Do you have some specific use-case in mind when those commands are needed?

In my experience restart doesn't sometimes work even with real hardware debugger, the connection is lost. And I never really missed it.

Detach I sometimes use to let the program run when disconnected from the debugger as opposite to the "Stop"button in the IDE which halts the CPU - the program is frozen. You wrote the monitor halt should detach the debugger, but shouldn't it stop the program? I must say I haven't debugged MCUs much those days so I may be missing something. I understand that even if I don't see the use for some commands they may prove useful for someone, so I would try to implement them if it seems possible.

Thanks

Jan

jdolinay, 2020-06-09:
Finally added the support for monitor halt and monitor reset - that is qRcmd,halt and qRcmd,reset. It's on branch support_qRcmd, for now only in the main sources in avr8-stub folder, not in the arduino library copy in arduino\library\avr-debugger\src folder.

I found out the halt/detach was already supported - the D packet so I just duplicated the functionality for qRcmd,halt packet. The stub removes any breakpoints and lets the program run. 
In fact, the qRcmd,halt is probably not needed because the debugger in VS Code always sends 'k' (kill) packet when the Stop button is pressed. And the stub does the same action for the k, D and qRcmd,halt packets.  So if you don't write any command into the pio_reset_run_target section in platformio.ini the behavior will be the same as if you write the **monitor halt** there.

The Reset button doesn't send anything by default, so it needs the **monitor reset** command in the pio_reset_halt_target section to do the reset. It seems to work but I'd like to test it more. The stub does the restart by watchdog reset as recommended by the AVR documentation. The debugger can only connect after the stub initializes, so the result is quite similar to what happens when you start new debug session - you see the target already running. I see one advantage - you can restart the program without uploading it again.  
If you get a chance try it and let me know what you think.

msquirogac, 2020-06-11:
I've forked the platform-atmelavr repository and pushed a [commit](https://github.com/msquirogac/platform-atmelavr/commit/1ca30d078038650fbcbd3c538757fef4f5a856a3) with full integration of avr-debugger to Platformio. To test it just use the following configuration on the **platformio.ini** file:
~~~
platform = https://github.com/msquirogac/platform-atmelavr.git#avr_debugger
board = nanoatmega328new
debug_tool = avr_debugger
debug_port = /dev/ttyUSB0
build_flags =
    -DAVR8_BREAKPOINT_MODE=1
~~~

Besides that, I submitted [this](https://github.com/msquirogac/pio-debug) simple yet useful library to PlatformIO and you can use it like this:
~~~
#include "pio-debug.h"

PIODEBUG(debug_init();)
// some other code
PIODEBUG(breakpoint();)
~~~
Now you can change between a release and debug build without having to change any line of code because the compiler will auto remove them for you based on the definitions that PlatformIO adds for both modes.

I want to test it a bit more before creating a PR, but with this little trick plus your latest commits the debug experience is pretty close to what you would get with an external debugger.

jdolinay, 2020-06-11:
Thanks for testing it and for the useful links. I will look at the platform-atmelavr and the library to understand how it works. 
In the meantime to the monitor commands...
Based on the link you sent I'd say there are 3 we should consider: run, halt, reset.
What they should do according to the doc of the SmartProbe:
**run** - let the target run, not removing the breakpoints. The effect is the same as the Continue (F5) button in the VS Code toolbar.
**halt** - stop the CPU.
**reset** - reset the target and disconnect the debugger.

Now I am not sure what they should do in our case. Here's what I think:
**run** - probably not needed, or it could be implemented as "detach" - to remove breakpoints and let the target run, as is now doing the halt. More on this below.

**reset** - leave as is now - restart the target. The debugger is disconnected as a side effect, it cannot communicate with our stub until the code of debug_init() is executed again.

**halt** - probably remove, don't support. It makes sense in combination with the reset command - the "monitor reset halt" to restart and stop the program at the beginning, but the problem is that the stub is part of the user program, not a separate hardware probe which can control the CPU from outside. So when we do reset we lose control and cannot get it back until the debug_init() is executed to configure the serial line etc. So in fact we cannot stop the program earlier than behind the debug_init() line.   
I can imagine simple implementation (with flash breakpoints only) which would insert breakpoint after the debug_init() call and restart. But this can be "emulated" by setting the BP by hand before doing Reset or calling breakpoint() in the code - as you wrote in the previous post. So it seems not needed.  
Then I can imagine implementation which would somehow initialize the stub before main() is executed and stop there, but this would be probably complicated (if not impossible); it seems not worth the effort.

So I think for the best user experience and minimal confusion in the stub code we could have the following behavior:
- When user clicks Restart, PIO sends monitor reset, we reset the target and it stops somewhere in the code when the stub re-initializes and responds to debugger again.
- When user clicks Stop, we don't define any commands in the platformio.ini and the debugger sends the kill packet automatically which is already supported by the stub and does "detach".
Or the PIO could send monitor run with the same effect - the stub removes breakpoints and lets the program run. 

What do you think? What should be the ideal behavior of the Restart and Stop buttons?

msquirogac, 2020-06-11:
Hi,
I'ven thinking something similar. About what you said here:

>Then I can imagine implementation which would somehow initialize the stub before main() is executed and stop there, but this would be probably complicated (if not impossible); it seems not worth the effort.

It is possible but not worth it. I found some ways [here](https://embeddedartistry.com/blog/2019/04/08/a-general-overview-of-what-happens-before-main/) by meddling with the `_start` function or by modding the bootloader a bit. Certainly not the most friendly things to do so I also think we should left it aside.

>When user clicks Restart, PIO sends monitor reset, we reset the target and it stops somewhere in the code when the stub re-initializes and responds to debugger again.

Totally agree and we could be sure of that behavior by creating a macro that calls the `debug_init()` , `breakpoint()` and `sei()`functions like this:

~~~c
#define debug_start() { \
	debug_init(); \
	sei(); \
	breakpoint(); \
	}
~~~

Then, the program execution would always stop at the instruction after `debug_start()` emulating that first stop while at the same time making things easier for the user to setup. That also wouldn't force things to the most advanced user which can still use the individual functions as they see fit.

We could also take the chance an implement that macro like this which only includes the calls when the user compiles for debugging. For reference review this [library](https://platformio.org/lib/show/7450/pio-debug)

~~~c
#define debug_start() { \
	PIODEBUG(sei();) \
	PIODEBUG(debug_init();) \
	PIODEBUG(breakpoint();) \
	}
~~~
With that, the user only has to add the call to `debug_start()` at the point where they want the debug session to start.

That macro, or something with the same effect, could be added to avr_debugger but as it has to work not only on PlatformIO but also on all the other ecosystems then the [condition](https://github.com/msquirogac/pio-debug/blob/0bce78be225f7ddfa404f9347426fbaf56628e29/src/pio-debug.h#L1) would have to check if **DEBUG** is defined.

Nonetheless, the pre-processor can still check for the **PLATFORMIO** and **\_\_PLATFORMIO_BUILD_DEBUG\_\_** definitions and define **DEBUG** if the former are defined, so the user doesn't have to meddle with the pre-processor.

Now, to avoid polluting the  [app_api.h](https://github.com/jdolinay/avr_debug/blob/master/avr8-stub/app_api.h "app_api.h") and [avr8-stub.h](https://github.com/jdolinay/avr_debug/blob/master/avr8-stub/avr8-stub.h "avr8-stub.h") headers another file could be created which defines that macro and also the conditional compiled version of the `breakpoint()` function.

>When user clicks Stop, we don't define any commands in the platformio.ini and the debugger sends the kill packet automatically which is already supported by the stub and does "detach".  

Yeah, maybe that would preferable so the memory footprint doesn't increase with redundant things, so better left it blank.
However, I've been thinking whether the `breakpoint()` could also be left without effect when the **kill** packet arrives, so not only the soft-breakpoints are released, the ones added on runtime, but also the hard-coded ones, which are all the calls to the `breakpoint()` function.

Probably a flag to conditionally execute the calls to `gdb_send_state()` and `handle_exception()` would do the trick, but I haven't tested it yet.

Regarding the new header file, once ready, I will create a PR for the DEBUG macro and all that conditional compiling I talked before and if you have no problem I will call the header `avr_debugger.h` to match the name used on the Arduino library and the debug tool in PlatformIO.

jdolinay, 2020-06-12:
Hi, I agree with you and I like the approach with the avr_debugger.h file. Also thanks for the link about the _start function, it's interesting.

So I will just leave the support for monitor reset and remove the monitor halt.

About removing the hard-coded breakpoints, yes, there could be a global variable and condition in the breakpoint() function like
 `if ( do_break ) call the debugger.. `

Then on kill we could set the variable to false and the breakpoints would be ignored - until reset which will reset the variable to the default value which must be to stop at the breakpoint. I think there is not much use for this feature. Perhaps to see what the program does if I let it run without breakpoints, but this I can do with one more upload and better, with the release built. 
What do you think?

msquirogac, 2020-06-12:
Glad you liked it.
About your question, yeah, it's just as you said, the main reason would be to see what the program does as if it didn't had any breakpoints. Besides that, another possible reason would be to comply better with the GDB documentation.
However, now that the fast release built feature was integrated, then I also think there is no much point in changing the breakpoint function, as it would only make it slower.

Now I have a question.
Time ago, I asked if this library would work without the Arduino framework, and certainly as you said it does. Nonetheless, I would like to know if it also can work on other models of the AVR family besides the ATmega328, ATmega1280, and ATmega2560. This for both the RAM and Flash breakpoint mode.

I ask these because the native [support](https://github.com/platformio/platform-atmelavr/compare/develop...msquirogac:avr_debugger) on PlatformIO is almost ready and I just need to add a line like [this one](https://github.com/msquirogac/platform-atmelavr/blob/5e8ee4e54345e78f8a7d1a35f5d5df116b93c28c/boards/nanoatmega328new.json#L19) to json file of all the supported boards. Yeah, that will take some time.

PlatformIO has a huge collection of supported boards and hardware and would be good if avr_debugger also worked on the other models of the AVR family.

jdolinay, 2020-06-14:
Right now only the ATmega328, ATmega1280, and ATmega2560 are supported. The support for RAM breakpoint mode wouldn't be hard, it requires defining the `RET_ADDR_MASK` plus maybe some other masks, then checking/defining the names of interrupt vectors which the stub uses and the code for serial communication (UART). 

Support for Flash breakpoints is a bit more complicated. Currently this project is designed for Arduino so it uses bootloader for writing to the flash memory. Now the question is if we support other AVRs, do we target other Arduino boards (I think there are not many more with AVRs) or do we target any boards with AVRs, including home-made? 

For other Arduinos we could use the same approach as in used now - I mean the stub communicating with a bootloader.

For any general boards there will most likely be no bootloader. For example, if I desing my own board with AVR I will use ISP programmer to upload the programs, bootloader would be just extra work and it may not even be available for the AVR I want to use... 
I think if we get outside the Arduino world which is quire unified, things can get complicated with lot of "variables".  

If we want to support also configurations without bootloader - programs uploaded via ISP - it would be in principle easier than what is used now (the bootloader API). The code to write to flash could be normal part of the library; just placed into the appropriate section of the memory. But adding this to the existing code could be complicated or confusing - probably some conditional compilation to either use bootloader API or include the flash-writing routines directly in the code. 

Other thoughts on this:
- I'm not sure if all the AVRs can rewrite their own flash - if it would be possible to support the flash breakpoints. 
- How many of those people who design their own hardware would want to use this debugger? If they work with AVRs, they will probably have some debug hardware like AVR Dragon and can use the real debugging via DebugWire. Actually, how many people still design their own hardware with "bare" AVR... it's so much easier and cheaper to put, for example, micro or Nano inside :) 
- Using this debugger requires serial interface or usb-to-serial chip. For a simple design with AVR it wouldn't be practical to add such interface just to be able to use the debugger.

So I think the usability of the debugger outside Arduino world is quite limited. What do you think?

msquirogac, 2020-06-14:
Well, you have a point on this one, nonetheless, maybe there is not much left to do but test, this if you only consider those models that are part of the same [family](https://en.wikipedia.org/wiki/Atmel_AVR_instruction_set#Optional_feature_instructions). The list can be reduced a bit more if you only consider the models that have the required peripherals as the external interrupt and watchdog.

Besides that, the [optiboot](https://github.com/Optiboot/optiboot) bootloader, which is also the used one by Arduino and the base for the one implemented on avr_debugger, already has support for a number of models outside those three you mention.

I think targeting Arduino users as the main public is a good idea at least for now. Once avr_debugger gets native [support](https://community.platformio.org/t/another-tool-for-debugging-avr-microcontrollers/13776/13?u=msquirogac) into PlatformIO it's popularity will rise and probably this same question will have to be asked again. Yet it would be good to have complete support for at least those three models on both RAM and Flash mode.

Talking a bit more about optiboot, I've recently found that a [PR](https://github.com/Optiboot/optiboot/pull/142) which adds support to flash programming from user code was added.
Isn't that the same what avr_debugger's bootloader does?

Maybe it would be worth checking it because as optiboot supports most of the AVR micro-controllers and said feature comes by default, then implementing the flash mode to any other model should be way easier to do.

jdolinay, 2020-06-14:
Thanks for another great link! I will look into the write-to-flash support in optiboot. Yes, it is exactly what the modified bootloader does. If this works "natively" with optiboot (and it looks that it should) it would really make things much easier. 
I agree with you the priority is complete support for at least the three models, which means adding flash breakpoints support for Mega - which should be easy if writing to flash can be done through optiboot.

msquirogac, 2020-06-15:
Hi,
I updated my fork of [platform-atmelavr](https://github.com/msquirogac/platform-atmelavr/commit/dcae3beda9d8790d7c837274a327c7d79dac0580) with support for all the boards with the atmega328 microcontroller.

These days, I've been wondering if the baud rate should be a fixed value or not and what should be the default value. Right now is [fixed](https://github.com/msquirogac/platform-atmelavr/blob/dcae3beda9d8790d7c837274a327c7d79dac0580/platform.py#L122) to 115200 which worked well for the atmega328, however, after reviewing the [code](https://github.com/jdolinay/avr_debug/blob/137443128aba5629e78b46e755317fa055d2b3cc/avr8-stub/avr8-stub.c#L78) related to the configuration of the UART port I found that not all the MCU models have the same default configuration.

Is there a reason to have 115200 baud on the atmega328 but 57400 baud on the atmega1280?
Wouldn't it be possible to have the same default for all of them?

Besides that, I found that some parts of the procedure used to configure the UART port can be improved. Even if this [equation](https://github.com/jdolinay/avr_debug/blob/137443128aba5629e78b46e755317fa055d2b3cc/avr8-stub/avr8-stub.c#L89) is technically correct it won't always return the best UBRR value for the requested baud rate, especially for "high speed".

For example, let's make F_CPU=16MHz and GDB_USART_BAUDRATE=115200
The used formula would reduce to **16.861**, however as the UBRR registers can only take the integer part, then **16** will be used, however, the nearest integer and probably best answer is **17**.

Now, let's take the inverse of that equation and see the true baud rate:
For UBRR = 16, it's inverse is **129032.25...** baud with a relative error of **12%**
For UBRR = 17, it's inverse is **121212.12...** baud with a relative error of **5.21%**
Therefore, the best configuration certainly is UBRR = 17, yet the programed code will use 16.

I propose to modify the procedure to calculate the UBRR register, which would naturally improve the stability of the system. After reviewing a bit of the available techniques I found this [manual](https://www.nongnu.org/avr-libc/user-manual/group__util__setbaud.html) that uses a header provided by Atmel themselves. What I like about it is that it already takes into account the "peculiarities" the hardware has and even takes into account the effect of the U2X0 bit.

msquirogac, 2020-06-15:
Hi,
I updated my fork of [platform-atmelavr](https://github.com/msquirogac/platform-atmelavr/commit/dcae3beda9d8790d7c837274a327c7d79dac0580) with support for all the boards with the atmega328 microcontroller.

These days, I've been wondering if the baud rate should be a fixed value or not and what should be the default value. Right now is [fixed](https://github.com/msquirogac/platform-atmelavr/blob/dcae3beda9d8790d7c837274a327c7d79dac0580/platform.py#L122) to 115200 which worked well for the atmega328, however, after reviewing the [code](https://github.com/jdolinay/avr_debug/blob/137443128aba5629e78b46e755317fa055d2b3cc/avr8-stub/avr8-stub.c#L78) related to the configuration of the UART port I found that not all the MCU models have the same default configuration.

Is there a reason to have 115200 baud on the atmega328 but 57400 baud on the atmega1280?
Wouldn't it be possible to have the same default for all of them?

jdolinay, 2020-06-17:
Hi, 
I am not sure now why the baudrate is 57400 for the Arduino mega with atmega1280. The old version of Nano also uses 57400.  
It is possible that I just used the same value as used by the bootloader for the board but I have a feeling that when I tried the Mega it didn't work with 115200. It may be related to the usb-to-serial chip used on the board which limits the speed. 

So I'd say there may be some boards for which there is "physical" requirement of the lower baud rate. But for most boards 115200 should work, so we could use 115200 as default for all and if someone needs lower baudrate, it can take some "advanced configuration". 
I am now trying the flash breakpoints with the standard optiboot bootloader and I think when I eventually get to implementing this for the Arduina Mega I should be able to say more about the  baud rate also.

msquirogac, 2020-06-17:
Yeah, we need a bit more testing to validate if 115200 is OK for most of the microcontrollers and boards. I think that the first deciding factor is the clock speed, for 115200 @ 16MHz the error is 2.1% which isn't the best but "works". However, for the same baud rate at 8MHz, the error goes up to 8.5%, which would make it very unstable, and lowering the baud rate to at least 57600 would be required.

One option would be to make 57600 the default one as it is more reliable, but the problem would still exist for low-speed clocks, like 4 MHz or less.
Another option would be to make the default value dependable on the F_CPU definition, then for clocks of 16MHz or more the default is 115200, and for everything else, the default is 57600. Even more, it would be better to add a third and fourth value for lower speed clocks, which is a usual setup for low power systems. And of course, the user should be able to change the value if needed.

What do you think?

jdolinay, 2020-06-18:
You are right about the problem with baudrates for lower clock speed. I think the solution which is used in the optiboot bootloader is good. It lets you define the baud rate by compiler directive and if you don't, it provides its own default based on F_CPU like this:
~~~c
/* set the UART baud rate defaults */
#ifndef BAUD_RATE
#if F_CPU >= 8000000L
#define BAUD_RATE   115200L // Highest rate Avrdude win32 will support
#elif F_CPU >= 1000000L
#define BAUD_RATE   9600L   // 19200 also supported, but with significant error
#elif F_CPU >= 128000L
#define BAUD_RATE   4800L   // Good for 128kHz internal RC
#else
#define BAUD_RATE 1200L     // Good even at 32768Hz
#endif
#endif
~~~
It also check the value and displays error/warning if the error is too high.
I'll put it on my todo list to update the UART code like this :) 
For now, for the PlatformIO definition of the debugger I think the 115200 would still be a good default as majority of people will be using this with Arduino boards and the modern boards like uno, nano, mega with 2560 will work with 115200.  I admit I have little idea how the definition of the  debugger for PIO which you do works, but I suppose the user will be able to define custom baudrate in the platformio.ini file if needed.

jdolinay, 2020-07-07:
Hi,
The 2 features we discussed are now working:
1. Baud rate can be defined by symbol AVR8_USER_BAUDRATE, e.g. AVR8_USER_BAUDRATE=9600. If not defined, defaults are used - this is the way used in the Optiboot bootloader.

2. Flash breakpoints can be used with the do_spm function from Optiboot bootloader. This means that flash breakpoints are now also available for Arduino Mega if the bootloader is changed to optiboot instead of the default one.

Unfortunately, using the Optiboot for writing to flash still requires that the users replace the bootloader in their board because the flash-writing function is available in Optiboot from version 8.0 and Arduino still uses version 4.4. So it's not as easy to use as I hoped. But still there is the advantage that the flash breakpoints can be easily implemented for any AVR supported by Optiboot (after the code in the avr-debugger is ported for that AVR; no extra work with porting the custom bootloader).  

Also loading via debugger is not available with the Optiboot option but that's not a problem for PIO which can automatically upload using avrdude when starting the debug session. 

I kept the option to use the custom bootloader for atmega328 for those who already use it and/or need the load feature. So users of atmege328 can choose between Optiboot or the avr_debugger bootloader. For Arduino Mega and any future supported boards I expect that only the Optiboot option will be available.

msquirogac, 2020-07-08:
That sounds amazing!!
I will update my fork of the atmelavr platform with the new baud rate defaults and create a PR so that avr_debugger becomes officially supported by PlatformIO.
Thank you