jdolinay/avr_debug #19: impossible to debug with platformIO and U8G2 lib

lienrag35, 2020-11-20:
Hello,
I succeeded to debug with this usefull library on atmega 2560 and platformIO on vscode using integration suggested by platformio but as soon as I add u8g2 lib or more code , linker shows these errors :
`Linking .pio\build\megaatmega2560\firmware.elf
AppData\Local\Temp\ccPR247p.ltrans1.ltrans.o: In function `breakpoint':
Documents\PlatformIO\Projects\TestDbg/.pio\libdeps\megaatmega2560\avr-debugger\avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
Documents\PlatformIO\Projects\TestDbg/.pio\libdeps\megaatmega2560\avr-debugger\avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
Documents\PlatformIO\Projects\TestDbg/.pio\libdeps\megaatmega2560\avr-debugger\avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
Documents\PlatformIO\Projects\TestDbg/.pio\libdeps\megaatmega2560\avr-debugger\avr8-stub/avr8-stub.c:2066: undefined reference to `regs'`

I tried multiple manaipulation included following the way you describe how ot integrate but I still have these errors. I have created aspecific github repo for the snippet code that shows the errors. [here](https://github.com/lienrag35/TestDbg)

Thank you for your help.

jdolinay, 2020-11-21:
Hi, I was able to reproduce the problem and what solved it was to remove the "static" keyword in the declaration of regs variable in the avr8-stub.c file. If you go to this file (it's in libdeps{avr-debugger folder in your workspace), find this definition:
static unsigned char regs[GDB_NUMREGBYTES];	/* Copy of all registers */
and just remove the static word so that it is:
unsigned char regs[GDB_NUMREGBYTES];	/* Copy of all registers */

Then the project should build without errors.
To tell the truth, I have no idea what is the cause of this error. The error was actually reported in inline assembly code, in function save_regs1 and others. I guess the compiler placed the variable into different code section where the linker could not find it or something like this... Removing the static keyword should do no harm.

lienrag35, 2020-11-21:
Hi

Perfect I just test and now I can debug all my program without using this serial.print .... :)

thank you very  much.

closed 2020-11-21 by lienrag35

maxgerhardt, 2020-11-21:
Indeed that's strange. Maybe GDB thinks the variable is unused and optimizes it away? Does using `unsigned char regs[GDB_NUMREGBYTES] __attribute__((__used__);` fix things too?

jdolinay, 2020-11-21:
I just tested it and it doesn't fix it. The error remains the same - linker reports "undefined reference to `regs'", for example in static inline void save_regs1 (void). There is inline assembly code which references the regs variable like this:
```
asm volatile (
				"push	r0	\n"
				"in		r0, __SREG__ \n"
				"cli	\n"			/* disable interrupts */
				"sts regs+32, r0\n"	/* save SREG to its place */
				"pop r0 \n"		/* restore r0 from stack */
				"sts	regs+31, r31\n"		/* save R31 */
				"sts	regs+30, r30\n"		/* save R30 */
				"ldi	r31, hi8(regs)\n"	/* Z points to regs */
				"ldi	r30, lo8(regs)\n"
```
The regs variable is used also in C code so the compiler should not think it's unused. 
I think it may be related to the fact that ATmega2560 has more than 64 KB of memory but actually that is the program memory; the data (RAM) address space is not larger than 64 KB. 
Static variables should only be visible in the same .c file which is true here, but maybe the assembly code is put into different compilation unit which does not see the variable, or something like this... But why would in happen only after the library U8g2 is inserted... Maybe the compiler starts to reorganize the code once there is a lot of it... Strange.

maxgerhardt, 2020-11-21:
Weird. Then making the variable global should be the way to fix it for now.. Maybe change the name from `regs` to something more "internally" sounding so that users don't accidentally collide with it?

jdolinay, 2020-11-22:
Yes, the name is not good. I'll put it on my todo list and rename it when I have some time to test it, so that it doesn't break the code.

Deepscorn, 2020-12-17:
Have same problem. Arduino Uno rev.3. PlatformIO, CLion, Windows 10, MinGW. Just adding/removing some call to my method makes error appear and disappear. @jdolinay Thank you very much for a workaround with removing static keyword.

jdolinay, 2020-12-20:
@Deepscorn Thanks for your report. If it happens on Uno too, then it's not related to the flash size of over 64 kB on Mega as I thought. Good to know...

vdavy, 2021-02-07:
Hi, I have the same problem.
Using CLion and platformio with my project https://gitlab.com/vincentdavy/thermo-lab

I use this LCD I2C management lib (working well) https://platformio.org/lib/show/576/LiquidCrystal_I2C (check the platformio.ini) and this is making me having the error reported in this issue, specifically here https://gitlab.com/vincentdavy/thermo-lab/-/blob/master/src/lcdmanager/LCDManager.h#L45

When I comment out this line, my project builds properly with the debug option. If the line is enabled, can't build having the error mentionned in this issue.
Here is the log :
```
Linking .pio/build/uno/firmware.elf
/tmp/ccAsg1nD.ltrans0.ltrans.o: In function `save_regs1':
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/tmp/ccAsg1nD.ltrans0.ltrans.o:/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2163: more undefined references to `regs' follow
collect2: error: ld returned 1 exit status
*** [.pio/build/uno/firmware.elf] Error 1
```

If any further details or action from my part is required, please fell free to ask.
Any help would really be apreciated.

PS : removing the static keyword make the project building ! 
Would you like a PR with this fix ?

Thanks

jdolinay, 2021-02-07:
Hi, thanks for the report. If the project builds OK with the static keyword removed, just remove it and all should be fine. It has no effect on functionality of the program.

I'm not sure what fix you mean, removing the static keyword? 
I would not want to just remove it in the code for all users, because using the static keyword is actually correct in this case and the name "regs" is not a good one, so without "static" it may cause conflicts if someone happens to use the same name if their code. So I plan to change the name to something more robust and then perhaps remove the static keyword if I cannot find out what causes the problem.

vdavy, 2021-02-08:
Hi, 

If you need somebody to test the reworked version, feel free to ask.

jdolinay, 2021-02-10:
Hi, 
Thanks. Could you try adding `lib_archive = false` to your platformio.ini file and check if the project builds without the error?

I think I understand the problem now. The PIO builds the libraries into archive files (.a) and then links these archives with your code. The static variable `regs` which is used in the breakpoint() function is then not visible and causes error. 

Another way to get rid of the error would be not to use the breakpoint() call in your program, but that's not comfortable.

vdavy, 2021-02-11:
Hello, it does not work for me.
If if add the `lib_archive = no` I can't build the project either in release and debug mode.
Here is the log in release mode :
```
Linking .pio/build/uno/firmware.elf
/tmp/ccgoHnhu.ltrans1.ltrans.o: In function `__vector_1':
<artificial>:(.text+0x370): undefined reference to `regs'
<artificial>:(.text+0x376): undefined reference to `regs'
<artificial>:(.text+0x37a): undefined reference to `regs'
<artificial>:(.text+0x37c): undefined reference to `regs'
<artificial>:(.text+0x37e): undefined reference to `regs'
/tmp/ccgoHnhu.ltrans1.ltrans.o:<artificial>:(.text+0x448): more undefined references to `regs' follow
collect2: error: ld returned 1 exit status
```

Here is the log in debug mode :
```
Linking .pio/build/uno/firmware.elf
/tmp/cc0myndu.ltrans2.ltrans.o: In function `save_regs1':
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2066: undefined reference to `regs'
/tmp/cc0myndu.ltrans2.ltrans.o:/home/vincent/dev/arduino/thermo-lab/.pio/libdeps/uno/avr-debugger/avr8-stub/avr8-stub.c:2163: more undefined references to `regs' follow
collect2: error: ld returned 1 exit status
```

If I comment out the `breakpoint();` call, it does not change anything. Build still failing.
Hope this can help.

maxgerhardt, 2021-02-11:
Can you check whether using a newer compiler versions makes a difference?

Add

```
platform_packages =
  ; use GCC AVR 7.3.0+
  toolchain-atmelavr@>=1.70300.0
```
to the `platformio.ini` and retry.

vdavy, 2021-02-11:
GG ! It worked for me project.
Well done, so platformio should upgrade the compiler version provided with AVR packages.

Here is my project link if you need to check something : https://gitlab.com/vincentdavy/thermo-lab

jdolinay, 2021-02-11:
I tried with the thermo-lab project and can confirm what @vdavy wrote. Using the newer compiler version solves the problem. No need to use the `lib_archive = false` option. Thanks @maxgerhardt !

In my simple test project the `lib_archive = false` did help, but not in the thermo-lab project. So it seems the problem was in the way the compiler handles access to static variables in inline assembly... who knows. Anyway, I think I can leave the `regs` variable as it is for now. Though I don't like the name...

maxgerhardt, 2021-02-23:
hey @vdavy, just FYI: I've edited my comment above regarding the `platform_packages` directive to get a newer toolchain. The version I originally posted did use the newer 7.4.0 GCC that solved the linking error, but we also found out in #24 that this expression leads to using that toolchain which has no avr-gdb -- debugging is impossible without that. The expression was corrected to get a good 7.4.0 toolchain with AVR-GDB, with which debugging is possible. So if you had problems, you should try again with the update.