jdolinay/avr_debug #20: MAC setup

mapicard, 2021-01-08:
Hi,
Setting up my environment according to your posts in VScode on a MAC works like a charm to upload to my UNO board but debugging does not work as it cannot find "avr-dbg" program. Does it need to be installed separately ?

tx

jdolinay, 2021-01-10:
Hi, yes, avr-gdb is part of the Atmel AVR toolchain - the tools for building program for Atmel AVR microcontrollers. For Windows it is usually included in the Arduino software, but in some versions they did not include it and in some version it does not work. In Linux avr tools are installed or can be easily installed separately. On Max I don't know but, perhaps you already have it if you have the Arduino IDE but you can also get the Atmel AVR Toolchain from here: https://www.microchip.com/en-us/development-tools-tools-and-software/gcc-compilers-avr-and-arm, see the "AVR 8-bit Toolchain 3.6.2 - Mac OS X 64-bit".

mapicard, 2021-01-15:
Thanks I'll give that a try.

mlewus, 2021-01-28:
Your library is trying to use the same serial port interrupt that is already assigned to the debugger. Wifi.h pulls in other header files, each of which pulls in still more headers. One of them, or your program, is trying to redefine/use the serial port. There is a section in the *terrific* avr_debug manual where the author talks about modifications you need to make to Arduino core files in order to prevent these kinds of conflicts.

jdolinay, 2021-01-28:
@mapicard sorry, for some reason I didn't get email notification about your last message so I only saw it now.  @mlewus is right, you need to prevent the compiler from including the definition of the serial port interrupt into your program. 

In VS Code on Windows it somehow works for me without doing anything, I just do not use the Arduino Serial in my code and there is no error. But maybe it's because I didn't use the Wifi library. In ideal case the compiler should not include the code of the Arduino definition of the interrupt vector if it is not used in your program, so if you can find and remove any reference to the Arduino Serial object in your code this could be the easiest solution.  

Another easy way to solve this would be to exclude the file HardwareSerial0.cpp from build, but I guess that's not possible in VS Code,  we have no control over the build process. 

So the last option is to modify the HardwareSerial0.cpp file. It's located in ...\arduino-1.8.13\hardware\tools\avr\avr\include\avr\ (on windows). You can just comment out the following block in the file:

#if defined(USART_RX_vect)
  ISR(USART_RX_vect)
#elif defined(USART0_RX_vect)
  ISR(USART0_RX_vect)
#elif defined(USART_RXC_vect)
  ISR(USART_RXC_vect) // ATmega8
#else
  #error "Don't know what the Data Received vector is called for Serial"
#endif
  {
    Serial._rx_complete_irq();
  }

The problem is that this change affects all your Arduino projects - Serial will not work. Still it may be easier to just comment it out  as needed than the "clean" solution below.

The "clean" solution would be to use conditional compilation to only exclude the block if the debugger is used like this:

#ifndef AVR_DEBUG 
#if defined(USART_RX_vect)
  ISR(USART_RX_vect)
#elif defined(USART0_RX_vect)
  ISR(USART0_RX_vect)
#elif defined(USART_RXC_vect)
  ISR(USART_RXC_vect) // ATmega8
#else
  #error "Don't know what the Data Received vector is called for Serial"
#endif
  {
    Serial._rx_complete_irq();
  }
 #endif

For this to work you need to define the AVR_DEBUG symbol for building your project with the debugger. Which is not easy with   Arduino build system :( But it should be possible in the same way as turning off the compiler optimizations, as described in the Step 3 in the manual in the VS code setup chapter. Just add -DAVR_DEBUG to compiler.cpp.flags in the platform.local.txt file. But I didn't try it...

closed 2021-03-02 by jdolinay