jdolinay/avr_debug #21: Debugger Hardware

mlewus, 2021-01-27:
I probably just missed it but I read the documentation and did not find any reference to the debugger hardware. Does this work with the ATSTK500 or does it require the more expensive ATATMEL-ICE debugger?
Thanks

mlewus, 2021-01-28:
I just reread the documentation and I think I understand what is  going on now. I don’t need any debug hardware,  the reserved pin int0 or int1 on the ‘328 is just to allow the driver to generate a software interrupt. If so please close this. Thanks

jdolinay, 2021-01-28:
Yes, it's correct, no debug hardware is needed and the int0 or int1 is used to generate software interrupt because there is no software interrupt instruction in the AVR CPU.

closed 2021-01-28 by jdolinay

mlewus, 2021-01-28:
Hey Jan,
I just wanted to say thanks for creating this and making it open source. I
can't tell you how many hours I've spent pulling my hair out with bugs that
are not easily found with Serial.print() on the '328

Mark Lewus

jdolinay, 2021-01-29:
Hey Mark,
Thank you for your comment. I'm glad to hear that. 
Jan Dolinay