jdolinay/avr_debug #24: Incompatibility with new toolchain

maxgerhardt, 2021-02-20:
A weird behavior, similiar to #19 is occuring. The last comments in that issue state how upgrading the AVR-GCC toolchain to 7.3.0 (from 5.4.0) using 

```
platform_packages =
  toolchain-atmelavr@>=1.70300.0
```
fixes the compilation error about 

```
undefined reference to `regs'
```

but, as soon I do that in the very minimal project as outlined in [here](https://community.platformio.org/t/how-to-debug-on-arduino-mega-2560/18446/2?u=maxgerhardt) with my Uno, debugging doesn't work anymore. There's an infinite loading bar going on and no breakpoint is being hit. Using the `breakpoint()` function also doesn't work.

`platformio.ini`

```ini
[env:uno]
platform = atmelavr
board = uno
framework = arduino
debug_tool = avr-stub
debug_port = \\.\COM14
lib_deps =
    jdolinay/avr-debugger @ ~1.1
platform_packages =
  toolchain-atmelavr@>=1.70300.0
```

with `src\main.cpp`

```cpp
#include "Arduino.h"
#include "avr8-stub.h"

void setup()
{
  // initialize GDB stub
  debug_init();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

doesn't work, infinite loading (blue bar chasing from left to right)

[image]

Removing 

```
platform_packages =
  toolchain-atmelavr@>=1.70300.0
```

makes it work.

[image]

So there's something magical going on with compiler bugs or linking errors or the assembly code or whatever so that this doesn't work on a new toolchain. 

Since hopefully PlatformIO will update the the 7.4.0 toolchain soon, it should be found out why this doesn't work on 7.4.0.

jdolinay, 2021-02-21:
I can confirm this, I am able to reproduce the problem.  
From my experiments it seems the problem may be in communication between PIO and the new avr-gdb rather than in the way the program is built. I tried the following:

- switch to the new toolchain with the `platform_packages` option
- build and upload the program to arduino.
- try to debug - it hangs without establishing connection with the board.
- disable the new toolchain (remove the `platform_packages` option) and try to debug **without re-uploading the program**.
- now the connection is established, the debugger is connected. It does not work properly but that is likely because the program in the board is built with different version of the toolchain than the avr-gdb expects.

So if I try to connect to the same program with the debugger of the new toolchain, it hangs. With the old toolchain it works. That's why  I suspect there is some problem in the way the avr-gdb is started or parameters are passed to it with the new toolchain.

Note: I use `empty debug_load_cmds =` option to prevent PIO from loading the program before debug.  

I am going to try it in eclipse with the new toolchain when I set it up...

maxgerhardt, 2021-02-21:
If it's a problem with the new `avr-gdb` then I'll try a crude trick -- replace the `avr-gdb.exe` file in the toolchain using the old toolchain.

jdolinay, 2021-02-22:
Could the problem be that the avr-gdb.exe is missing in the new PIO toolchain? 
If I look into the bin directory of the new toolchain (c:\Users\jan\.platformio\packages\toolchain-atmelavr\bin\) it is not there. In the old toolchain avr-gdb.exe is there.

I tested it in Eclipse building with the PIO toolchain (avr-gcc 7.3) and using avr-gdb from the latest Microchip toolchain it works OK. I think it would be better to use the same/compatible version of avr-gdb and the avr-gcc just in case. By the way, the Microchip toolchain seems pretty outdated, it has avr-gcc version 5.4 (2015) and avr-gdb version 7.8. (2014).

maxgerhardt, 2021-02-22:
> If I look into the bin directory of the new toolchain (c:\Users\jan.platformio\packages\toolchain-atmelavr\bin) it is not there. In the old toolchain avr-gdb.exe is there.

Aha okay there's something weird going on. 

When direct PlatformIO to use the new toolchain, it gets me `toolchain-atmelavr` with version code from the `package.json` as `2.70300.201015`. There is indeed *no* `avr-gdb.exe` in the `bin/` folder there.

But then there's also the fodler `toolchain-atmelavr@1.70300.191015` that *does* contains `avr-gdb.exe`, datecode 12th march 2019.  So the .201015 (2020 Oct. 15th) package is missing avr-gdb. Now let me try with 

```
platform_packages =
  toolchain-atmelavr@1.70300.191015
```
instead..

maxgerhardt, 2021-02-22:
I've opened https://github.com/platformio/platform-atmelavr/issues/242  for this issue in the PlatformIO atmelavr platform to have this investigated.

maxgerhardt, 2021-02-23:
Closing since it's handled in the linked issue and solution exists and is understood.

closed 2021-02-23 by maxgerhardt