jdolinay/avr_debug #25: Using Alternate Serial Ports

diasrobson, 2021-03-14:
Hi There just wanted to say amazing job with this project, it makes my life so much easier.

so I'm using an atmega2560 with serial ports to spear unfortunately due to how my PCB is it can’t operate without the main serial port. so I was wondering if there was a way to change the debugger to use one of the other Usart lines and if so what line would I need to edit

jdolinay, 2021-03-14:
Hi, thank you! 
It should be quite easy to change it. Probably you just need to change the number in the USART register names, for example, UCSR0A to UCSR1A, and so on. All the code dealing with USART is in the avr8-stub.c file in these functions:
uart_init
getDebugChar
putDebugChar

And also update these two macros:
#define	UART_ISR_VECTOR	USART0_RX_vect
#define	UART_RXINT_PENDING()  (UCSR0A & (1<<RXC0))

I hope it will work.

diasrobson, 2021-03-14:
Thanks @jdolinay, It worked perfectly.

Just for reference for other people, Here are the changes I've made to get it working with Serial Port 3 on the atmega2560/Arduino Mega

`#define	UART_RXINT_PENDING()  (UCSR3A & (1<<RXC3))`

`#define	UART_ISR_VECTOR	USART3_RX_vect`

```
/* ---------- UART communication routines  ------------- */

/* Initialize UART */
static void uart_init(void)
{
	/* Init UART */
	UCSR3A = _BV(U2X3);		/* double UART speed */
	UCSR3B = (1 << RXEN3 ) | (1 << TXEN3 );		/* enable RX and Tx */
	UCSR3C =  (1 << UCSZ30 ) | (1 << UCSZ31 ); /* Use 8- bit character sizes */
	UBRR3H = ( GDB_BAUD_PRESCALE >> 8);
	UBRR3L = GDB_BAUD_PRESCALE;
	UCSR3B |= (1 << RXCIE3 ); /* Enable the USART Recieve Complete interrupt ( USART_RXC ) */

}

/* Read a single character from the serial port */
static uint8_t getDebugChar(void)
{
	/* wait for data to arrive */
	while ( !(UCSR3A & (1<<RXC3)) )
		;

	return (uint8_t)UDR3;
}

/* Write a single character to serial port */
static void putDebugChar(uint8_t c)
{
	/* Wait for empty transmit buffer */
	while ( !( UCSR3A & (1<<UDRE3)) )
		;

	/* Put data into buffer, sends the data */
	UDR3 = c;
}
/* ---------------- end UART communication routines ---------------------------------- */
```

closed 2021-03-14 by diasrobson

jdolinay, 2021-03-16:
Great! Thanks for sharing your solution!

maxgerhardt, 2022-03-07:
Someone reimplemented that again at https://community.platformio.org/t/is-it-possible-to-modify-avr8-stub-source-code-to-debug-arduino-mega-2560-while-keeping-serial-port-0-printing-functionality/26268/3?u=maxgerhardt

Can we introduce some macros to to control which UART registers and thus `Serial` is used? This would make the serial port selection easier.

jdolinay, 2022-03-07:
Yes, this seems like a good idea. I will do it.

jdolinay, 2022-03-13:
Done. It is now possible to select the port by defining AVR8_UART_NUMBER with the number of UART to be used. Example for PIO to use UART3: 
```
debug_build_flags =  
    -DAVR8_UART_NUMBER=3
```
If not defined, UART0 is used, as it was before.

CyberDNIWE, 2022-09-20:
Hi, I am using the version with AVR8_UART_NUMBER define to no avail.
My goal is similar: to use plain old Serial (USART0) as is for upload and communication, while simultaniously using Serial3 (USART3) for debugging. The setup is also fairly similar: the board with multiple serials (the one I'd like to debug) is Arduino Mega but instad of using some fancy USB-to-USART tool, I'm using Arduino Uno with its RESET shorted to GND and TX connected to Mega's TX3, and RX to Mega's RX3. Mega is on COM5 and Uno is on COM9.
I have tested that I can see serial output from Uno that way, however when I try to debug via it I get the error as if COM9 (which is Uno) is non existing.

```
========================= 1 succeeded in 00:00:04.756 =========================
Reading symbols from c:\Repos\Arduino\Mega2560DebugTest\.pio\build\megaatmega2560_DEBUGG\firmware.elf...
done.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = avr-stub
PlatformIO: Initializing remote target...
.pioinit:13: Error in sourced command file:
COM9: No such file or directory.
```
My platformio.ini looks like this:
```
[env:megaatmega2560_DEBUGG]
platform = atmelavr
board = megaatmega2560
framework = arduino
upload_port = COM5
monitor_port = COM5

debug_tool = avr-stub
debug_port = COM9

lib_deps = jdolinay/avr-debugger@^1.5
```
I have not tried to change avr-debugger sources as described above, so I can't say for certain if the problem lies within it or if it's the uno that messes it up somehow, so any help would be much appreciated.
(And I've also tried to swap RX->TX and vice versa to the same result)

CyberDNIWE, 2022-09-20:
As it usually happens, I've figured it out a few minutes after the post...
So just in case someone else finds themselves in the same situation the solution is as follows:

- Pins go 1-to-1, so TX3<->TX and RX3<->RX; with the secon board RESET shorted to its GND.
- If using platformio, use `build_flags` to define `AVR8_UART_NUMBER` inside `platformio.ini` insetad of `#define`-ing them in the beginning of `main.cpp`
- If presented with succesfully completed upload but immediately after `COM9: No such file or directory` (COM number is system specific) ensure that neither of your boards USBs come through the same dongle.