jdolinay/avr_debug #33: Missing breakpoints when using wire library

HeinzMu, 2021-04-19:
Hi Jdolinay, for a while I'm using your avr_debug tool and it's really great. But now I tested it with i2c communication and sometimes my breakpoints are ignored. Even the coded breakpoint function doesn't work sometimes. Do you have any idea for a workaround?
I'm using Arduino Mega 2560.

jdolinay, 2021-04-19:
Hi, thank you. 
One reason could be that there are too many breakpoints. If you use RAM breakpoints mode (default with the -DAVR8_BREAKPOINT_MODE=1) you should be able to set 7 breakpoints. If you are using the flash breakpoints mode (-DAVR8_BREAKPOINT_MODE=2) there are 3 breakpoints possible but there is bug in the code that could lead to something like this if you remove and then insert several breakpoints at a time. Quick workaround  for this case would be to open the avr-stub.h file and change the definition of AVR8_MAX_BREAKS from 4 to, for example, 8. 

If you are using RAM breakpoints or if the above does not help, please post your code (ideally some minimal program that shows this problem) and your platformio.ini so that I can try to reproduce it.

Best,
Jan

HeinzMu, 2021-04-20:
Hi Jan, thank you for your quick reply. Give me some days to try all your hints. You'll get a report about my results.

Best regards,
Heinz

HeinzMu, 2021-04-22:
Hello Jan, I tested your proposals. Unfortunatelly I got no solution. So I prepared a zip for you where all required files are attached. Unzip it and put it to the workspace of platformIO.
In release mode it'll run on a Mega2560 and the serial output is on default (Serial).
In debug mode the serial output is lead to Serial3.
When I start in degub mode it runs until line 123 of wiring.c . When pressing continue it stops at line 173 in main.cpp after the hard coded preakpoint. When I press 'Step over'  the program runs again and won't come back. Pressing 'Pause' it usually stops in twi.c between line 315 or 316. I suppose its a problem with the micros() function.
If you have any question feel free to contact me.

Best regards, Heinz

[RL_5x2_MUX-Test_Jan.zip](https://github.com/jdolinay/avr_debug/files/6356471/RL_5x2_MUX-Test_Jan.zip)

jdolinay, 2021-04-22:
Hello Heinz, Thank you for sending the project. I tried it but I was not able to reproduce the problem - I should have thought about that - without the same hardware there is not much chance the program would behave the same for me as for you.
Anyway, I can try some suggestions:

1. You could try using newer version of the avr/debugger. I think the ~1.1 option in your platform.ino will only take updates with 1.1.x while the library is now at 1.4. You could try ^1.1 which should get the new version (if I understand the PIO versions correctly). Anyway, this will probably not help, there were no big changes...

I don't understand why breakpoints should be missed, but I can say that probably it will not be possible to reasonably debug your program in the default RAM breakpoints mode. The Wire library uses interrupts and those are blocked when you debug the program, even if you click the continue button and let the program run. So the library will not be able to send/receive data. It is most likely also the reason why you find the program in the twi.c file around line 315. In that loop the code waits for variable twi_state to be changed in interrupt service routine, which will never happen. Normally the timeout would end this wait but the micros function is also not working because it depends on timer interrupt - so the program hangs there. 

The only situation when other interrupts are not blocked and the program should work is if you remove all breakpoints and let the program run (the hard-coded calls to breakpoint() are fine, they can stay). Then other interrupts should work and the program should work normally. The limitation is that you can only stop it by the pause (suspend) button, so it will stop at random moment. But maybe this can still be useful to some extent. 

2. To get much better debugging  experience you could use the FLASH breakpoints mode. The option is simple to set in PIO, just add -DAVR8_BREAKPOINT_MODE=2 to your build_flags. But you need to replace the bootloader in your Arduino with optiboot bootloader of version 8 or newer. If you download the complete code from this repository, you can find the .hex file for this bootloader in avr_debug\arduino\hardware\avr-debugger\avr\bootloaders\atmega2560\. There is some documentation on the process in the avr_debug.pdf in the How to enable breakpoints in flash memory chapter.

Best regards, 
Jan

HeinzMu, 2021-05-02:
Hello Jan, I found a solution for my problem so I'll close this issue. FLASH breakpoint mode I'm going to try later. Thank you very much for your support. 

Best regards, Heinz

closed 2021-05-02 by HeinzMu