jdolinay/avr_debug #34: Missing avr-gdb.exe in Arduino IDE version 1.8.13

darkless47, 2021-04-27:
Hi guys,
Greatly appreciate! The tutorial is extremely detailed and just helped me go through all the processes to make it work.

However, I think the latest Arduino IDE version which is 1.8.13 does not include the "avr-gdb.exe" in Arduino\hardware\tools\avr\bin\ folder.
Here is a link to a post related to this issue on the Arduino forum: https://forum.arduino.cc/t/missing-avr-gdb-exe-avr-core-1-8-3-included-in-ide-1-8-13/664566

The Arduino IDE did not update for almost a year and this issue remains. To solve it, just simply download the ZIP version (instead of the windows installer) of the 1.8.12 version of the Arduino IDE and copy&paste the "avr-gdb.exe" to the right place.  
Then it should work fine.

Just trying to help those people who downloaded the 1.8.13 Arduino IDE.
Thank you!

jdolinay, 2021-04-27:
Hi, Great! Thank you!

geetee24, 2021-12-29:
it is not in the current zip.  where can we find it?

geetee24, 2021-12-29:
here is latest file in the zip

https://downloads.arduino.cc/arduino-1.8.11-windows.zip

jdolinay, 2022-01-03:
Hi. I'm not sure if you mean that in the 1.8.11 there is the avr-gdb.exe or you need it and it is not there. 
According to the original message above the zip file for 1.8.12 should contain the file but I did not check it now.
I just checked the latest zip 1.8.19 and the avr-gdb.exe is not there :(  

There is a long history of this file missing of (perhaps worse) being included but not working because of missing dependencies. The Arduino IDE does not use it so I have little hope they will fix this. 

Personally I now use PlatformIO (PIO) with VS Code instead of the Arduino Extension. PIO does not depend on Arduino IDE.  It is easier to use the debugger in PIO and it's also easy to configure the build options, such as disabling compiler optimizations.