jdolinay/avr_debug #36: Linker error

tig, 2021-06-30:
I'm getting this:

```
Building in debug mode
Linking .pio\build\megaatmega2560\firmware.elf
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_25'
avr8-stub.c.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\megaatmega2560\firmware.elf] Error 1
```

My `platform.io`:

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[paltformio]
default_envs = megaatmega2560
description = TV Slider

[common_env_data]
build_flags = 
	-D __ASSERT_USE_STDERR
lib_deps_builtin = 
	SPI
	Wire
lib_deps_external = 
	arduino-libraries/Ethernet@^2.0.0
	marcoschwartz/aREST@0.0.0-alpha+sha.0e6b782568
	adafruit/Adafruit SleepyDog Library@^1.4.0
	andrealombardo/L298N@1.1.0
	adafruit/Adafruit_VL53L0X@^1.1.0
	adafruit/Adafruit_VL6180X@^1.2.0
	adafruit/Adafruit GFX Library@^1.10.7
	adafruit/Adafruit SSD1306@^2.4.3
	adafruit/Adafruit BusIO@^1.7.3
	bblanchon/ArduinoTrace@^1.2.0
	sparkfun/SparkFun VL53L1X 4m Laser Distance Sensor@^1.2.9
	sparkfun/SparkFun I2C Mux Arduino Library@^1.0.1
	bblanchon/ArduinoJson@^6.18.0
	thijse/ArduinoLog@^1.1.1
	jonblack/arduino-fsm @ ^2.2.0
	sparkfun/SparkFun Qwiic Relay Arduino Library@^1.2.0
	chris--a/Keypad@^3.1.1

[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
monitor_speed = 57600
debug_tool = avr-stub
debug_port = SERIAL_PORT
build_flags = 
	${common_env_data.build_flags}
lib_deps = 
	${common_env_data.lib_deps_builtin}
	${common_env_data.lib_deps_external}
	jdolinay/avr-debugger@^1.4
```

In `main.cpp`:

```c++
// Libraries
#include <Wire.h>
#include <ArduinoLog.h>
#include <assert.h>
#include "avr8-stub.h"
//#include "app_api.h" // only needed with flash breakpoints

...

void setup(void) {
  debug_init();

...

```

Please help. Thanks.

felias-fogg, 2021-06-30:
Hi, it looks as if you are using _Serial_ on the ATmega2560 in your program. This is not allowed, when you want to use the debugger.

tig, 2021-06-30:
> Hi, it looks as if you are using _Serial_ on the ATmega2560 in your program. This is not allowed, when you want to use the debugger.

Forgive my ignorance or if this is a stupid question.... 

Does this mean I need to remove all references to `Serial` in my program?

closed 2021-06-30 by tig