jdolinay/avr_debug #38: debug_message() is needlessly restrictive in only sending 47 chars

tig, 2021-07-01:
Related to #37 

`debug_message()`, for some reason, was implemented such that it only sends the first 47 chars of `msg`.

It seems this was done so it could utilize `gdb_ctx->buff` as a temporary buffer Which is `uint8_t buff[AVR8_MAX_BUFF+1];`.

However, I've proven that the underlying API that sends the text: 

```c
static void putDebugChar(uint8_t c);	/* Write a single character to serial port */
```

...has no such limitation. 

See `gdb_send_buff()`:

```c
__attribute__((optimize("-Os")))
static void gdb_send_buff(const uint8_t *buff, uint8_t sz)
{
	uint8_t sum = 0;

	putDebugChar('$');
	while ( sz-- > 0)
	{
		putDebugChar(*buff);
		sum += *buff;
		buff++;
	}

	putDebugChar('#');
	putDebugChar(nib2hex((sum >> 4) & 0xf));
	putDebugChar(nib2hex(sum & 0xf));
}
```

There's no reason `debug_message()` couldn't be re-implemented using the same technique.

Note also, that it's unnecessary that `debug_mesasge()` appends an `\n`. GDB automatically puts every `$0xxx#` command on a separate line.

jdolinay, 2021-07-02:
I agree. I don't remember why I used the temporary buffer to prepare the message instead of encoding it 'on the fly'.  Anyway, the 46 characters limit does not seem that bad to me to be worth rewriting the code. Please see also my comment at issue #37.

> Note also, that it's unnecessary that debug_mesasge() appends an \n. GDB automatically puts every $0xxx# command on a separate line.

I think the \n is there to tell GDB to output the message. I read somewhere that GDB will queue the messages untill \n is send and only then output the message to the console. But I didn't test it, so it may not be needed.

tig, 2021-07-03:
I concur with focusing on #37 and leavning debug_message alone.

Note that in my testing GDB does NOT wait for the `\n`; but the point I moot given we’re not touching `debug_message()`.