jdolinay/avr_debug #43: including avr debug breaks Serial.println etc in VC

geetee24, 2021-12-29:


jdolinay, 2022-01-03:
Unfortunately, the Serial cannot be used together with the debugger. The debugger uses the UART (serial communication) module, so Serial.print etc. will not work.

If you really need to print some messages while debugging you can use debug_message function which is part of the debugger library. It will output the messages to debugger console. Otherwise it is best to use conditional compilation to remove all Serial code from the program in debug version. 

For more info please see the avr_debug.pdf in the doc folder, the Important limitations of the debugger section.

RioTrax, 2025-01-03:
Hello, @jdolinay, first thanks a lot for that avr-stub library. 
I'm using arduino uno, so I'm obliged to use one unique Serial port for debug and for sending data to Serial Monitor (those data will be used later by java application....) .

I'm trying to solve the issue by using the SoftwareSerial library and and UART/USB converter....
So the native USB/serial COM4 will be used for debug and upload, and the ''virtual" COM5 for printing data.

**But it's not working. could you advise please!? Thank you in advance for your support**

When I start debug I'm getting errors:

> C:\Users\USERxxx\AppData\Local\Temp\ccWJUKxe.s: Assembler messages:
> C:\Users\USERxxx\AppData\Local\Temp\ccWJUKxe.s:2584: Error: register r24, r26, r28 or r30 required
> C:\Users\USERxxx\AppData\Local\Temp\ccWJUKxe.s:2673: Error: register r24, r26, r28 or r30 required
> lto-wrapper.exe: fatal error: avr-g++ returned 1 exit status
> compilation terminated.
> c:/users/riadh trabelsi/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
> collect2.exe: error: ld returned 1 exit status
> [.pio\build\uno\firmware.elf] Error 1

 **Here the configuration of my platformio.ini**
```[env:uno]
platform = atmelavr
board = uno
framework = arduino

monitor_port = COM5
monitor_speed = 115200

debug_tool = avr-stub
debug_port = COM4
upload_port = COM4
build_flags = -DAVR8_BREAKPOINT_MODE=1

lib_deps = 
    featherfly/SoftwareSerial@^1.0
	jdolinay/avr-debugger@^1.5
```
 **for illustration, here simple basic code :**

```
#include <Arduino.h>
#include "avr8-stub.h"
#include <SoftwareSerial.h>

#define Rx 10
#define Tx 11

SoftwareSerial FTDI(Rx, Tx); //COM5

int i=0;
void setup() {  
  FTDI.begin(115200);  
  FTDI.println("FTDI UART com Setup complete");  
  debug_init();  
  sei();
  pinMode(13, OUTPUT);  
}

void loop() {	
  i++;
  FTDI.println((String) i +  "message from FTDI!");
 	
  digitalWrite(13, HIGH);  
	delay(500);
	digitalWrite(13, LOW);
	delay(1000);
}
```

jdolinay, 2026-01-05:
> Would debug_message work on an arduino nano ? is it somehow possible to debug an avr target (without usb serial connection) from say an arduino nano host ? possibly something that you can upload to run on the target avr ? would it require connectin debugging pins (miso/mosi/sck ?) to arduino nano ?

Hi, debug_message should work on Arduino Nano if it uses the ATmega328, which the standard Nano does. but this applies to "normal" use with PC host running GDB. For the 2nd question about debugging from Arduino Nano as a host, simple answer is no :(  Maybe I do not understand your question, but this debugger works only over serial line and it communicates with GDB running on a PC. The debugging pins are not used at all. Using another Arduino as a host would be completely different thing. Maybe you mean using another Arduino Nano as a "debug probe" - in this case take a look at https://github.com/felias-fogg/dw-link.