jdolinay/avr_debug #44: Strange Memory Address

technical-publishing, 2022-01-25:
Hello, I am just new to avr_debug and I am using it in Visual Code with PlatformIO. It works fine for the first tests, but I have a question about the memory addresses. I have an Arduino UNO board attached and I want to check some String handling. The debugger shows address 0x800240 for the string, but the Atmege328 hasn´t such an address for SRAM. Where is my thinking error??

felias-fogg, 2022-01-25:
That is how avr-gdb treats the problem that the AVRs have separate data and program memory.  Program memory starts at 0x0 end extends theoretically until 0x7FFFFF. SRAM memory starts at 0x800000 (corresponding to real SRAM address 0x0). Note, that is what is displayed. The MCU internal registers see, of course, the value without the  0x800000.

technical-publishing, 2022-01-26:
Hi Bernhard, thanks a lot for your fast response and the well-understandable explanation.