jdolinay/avr_debug #45: a method for mcu debugger only using UART ports

wanggaoteng, 2022-04-26:
Hello, I find that there is a method only using uart ports: https://github.com/adamgreen/mri
Could you port it for avr (arduino)?
Best regards.

jdolinay, 2022-04-27:
Hello, if I understand the readme correctly, MRI is doing the same thing for ARM that avr_debug is doing for AVR (arduino). You could say the same thing they say about MRI about avr_debug: it is "debug monitor which allows the GNU debugger, GDB to debug AVR processors... " and it also uses only uart. 
Could you be more specific what you mean by porting MRI for avr? What features would you want in avr_debug that are in the MRI?
Best regards

wanggaoteng, 2022-04-27:
> Hello, if I understand the readme correctly, MRI is doing the same thing for ARM that avr_debug is doing for AVR (arduino). You could say the same thing they say about MRI about avr_debug: it is [...]

Hi, thanks for your great work, it's my fault that didn't realize avr-debug is the same with mri.
Best regards.