jdolinay/avr_debug #46: Impossible debugging VSCode PlatformIO avr-debug and Apple Silicon M1

VictorLasarteExo, 2022-05-17:
Good morning,

I usually work with VSCode and PlatformIO for my arduino projects.

I have use avr_debug before with VSCode, PlatformIO and Windows 10 and 11 without any problem. But it's imposible for me to make it work with my MBP with Apple Silicon.

The debugger never starts and the error I get in the debug console is:

undefineddyld[4130]: Library not loaded: /System/Library/Frameworks/Python.framework/Versions/2.7/Python
Referenced from: /Users/xxxxx/.platformio/packages/toolchain-atmelavr/bin/avr-gdb
Reason: tried: '/System/Library/Frameworks/Python.framework/Versions/2.7/Python' (no such file),           '/Library/Frameworks/Python.framework/Versions/2.7/Python' (no such file)

There is not Python 2.7 for Apple silicon, and installing it with brew under Rosetta 2 don't solve the problem.

The software I'm using is:
MacOS Monterey 12.4
VSCode 1.67.1
PlatformIO Core 6.0.0 and Home 3.4.1
avr-debugger 1.5

The hardware:
MacBook Pro M1
Arduino Mega 2560

Am I the only one with this problem? 

Thank you in advance.

jdolinay, 2022-05-19:
Hello,

I have no experience with Mac, but from the error messages the problem seems to be in avr-gdb and is related to Python. 
In the avr_debug.pdf there is short note at the end 'Some notes for Mac users' written by  @felias-fogg which could help you. Here is a quote from the note:

> One should point out that avr-gdb appears to be a bit 
problematical (that is in April 2021). It is not part of the toolchain in the Arduino IDE. PlaformIO 
contains avr-gdb, but the version does not have Python support. Then there is the homebrew 
version. However, even this version has a problem, since it cannot deal with the elf32-avr object file 
format (see: https://github.com/osx-cross/homebrew-avr/issues/216). The workaround described by 
xhero in this github issue worked quite well.

jdolinay, 2022-06-02:
This seems to be related either to Mac or to VS Code/Platform IO. The only idea I have is that it could be related to the message  asking if I trust the developers which started to show in newer versions of VS Code when you open a folder. Perhaps this could be the result of "un-trusted" folder and perhaps the trust/not trust question was not shown for Platform IO project and it just defaulted to not trusted... 

Anyway, what could help is to search google/Platform IO/VS Code forums for this problem.