jdolinay/avr_debug #48: Is there a problem with using COM port bigger than 9?

emaayan, 2022-12-23:
anything i've configured my com ports for arduino nano to be anything bigger than 9 

(i.e 10, 17 etc..) i get (if for example i'm using COM 10 )
when i try to debug 

.pioinit:13: Error in sourced command file:
COM10: No such file or directory.

[env:nanoatmega328new]
platform = atmelavr
board = nanoatmega328new
framework = arduino
monitor_port = COM10
upload_port = COM10
debug_port = COM10
debug_tool = avr-stub
monitor_speed = 115200
lib_deps = 
# 	mathertel/OneButton@^2.0.3
#	arduinogetstarted/ezLED@^1.0.0
 	marcoschwartz/LiquidCrystal_I2C@^1.1.4	
	jdolinay/avr-debugger@^1.5

ccrause, 2022-12-24:
For ports larger than com9, one has to specify as **\\\\.\\com10**, see [this Windows HOWTO](https://support.microsoft.com/en-us/topic/howto-specify-serial-ports-larger-than-com9-db9078a5-b7b6-bf00-240f-f749ebfd913e). Ideally the user software (seems like platformIO in your case) should ensure that the "\\\\.\\" prefix is added to all bare comport names on Windows to be on the safe side.

emaayan, 2022-12-24:
thanks, i actually switched to something bigger than 10 as i was troubleshooting, might wanna add it to an FAQ.

closed 2022-12-24 by emaayan