jdolinay/avr_debug #52: Help me with problem of debugging in Pio, plsssss

DHVBKU, 2023-06-27:
Hi admin, I'm new in using debug with arduino code. Because of my immaturity, allow me to ask about my problem:
I want to debug with Pio on VS code platform and I followed your instructions and fully loaded the latest library but I keep getting errors.
Let me know if I missed something or help me to fix it. Thanks
This is my code :
` #include <Arduino.h>
  #ifdef DEBUG_AVR
  #include <avr8-stub.h>
  #include <app_api.h>
  #endif
  #include <RFID.h>
  #include <SPI.h>
  #include <LiquidCrystal_I2C.h>
  #include <Wire.h>
  RFID rfid (10, 9);
  LiquidCrystal_I2C lcd(0x27, 20, 4);
  int nutbam1  = 6;
  int nutbam2  = 5;
  int nutbam3  = 4;
  int biendemactive   = 0;
  int biendemapp      = 0;
  int biendemmenu     = 0;
  int biendemcontrol  = 0;
  int biendemcontent  = 0;
  int biendemexit     = 0;
  int biendemquatrinh = 0;
  int serNum0;
  int serNum1;
  int serNum2;
  int serNum3;
  int serNum4;
  //ON-OFF
  void on() {
    lcd.clear();
    lcd.backlight();
    lcd.setCursor(6, 0);
    lcd.print("XIN CHAO");
    lcd.setCursor(3, 2);
    lcd.print(" CARD");
    lcd.setCursor(12, 2);
    lcd.print(" MENU");
  }
  // MÀN HÌNH CHÍNH
  void mainscreen() {
    if (biendemapp % 2 == 1) {
      lcd.clear();
      lcd.setCursor(6, 0);
      lcd.print("XIN CHAO");
      lcd.setCursor(3, 2);
      lcd.print(">CARD");
      lcd.setCursor(12, 2);
      lcd.print(" MENU");
    }
    if (biendemapp % 2 == 0) {
      lcd.clear();
      lcd.setCursor(6, 0);
      lcd.print("XIN CHAO");
      lcd.setCursor(3, 2);
      lcd.print(" CARD");
      lcd.setCursor(12, 2);
      lcd.print(">MENU");
    }
  }
  // MÀN HÌNH ỨNG DỤNG
  void applicant() {
    if (biendemapp % 2 == 1) {
      lcd.clear();
      lcd.print("THE CUA BAN");
    }
    if (biendemapp % 2 == 0) {
      lcd.clear();
[... 256 lines cut ...]
  File "C:\Users\LENOVO\.platformio\penv\lib\site-packages\semantic_version\base.py", line 647, in __init__
    self.clause = self._parse_to_clause(expression)
  File "C:\Users\LENOVO\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1043, in _parse_to_clause
    return cls.Parser.parse(expression)
  File "C:\Users\LENOVO\.platformio\penv\lib\site-packages\semantic_version\base.py", line 1063, in parse
    raise ValueError("Invalid simple block %r" % block)
ValueError: Invalid simple block '`'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
[image]
[image]
`

jdolinay, 2023-06-28:
Hi, This looks like some basic problem with PIO configuration. The instructions in my docs may be out of date. Please try the official PIO instructions here https://docs.platformio.org/en/stable/plus/debug-tools/avr-stub.html. If that does not help you can try to ask in PIO forum.

DHVBKU, 2023-06-28:
I get the problem is in the command line lib_deps =
    jdolinay/avr-debugger @^1.5 because when I remove this command my OUTPUT doesn't crash, but I still can't debug
[image]
[image]