jdolinay/avr_debug #53: Can't debug on Mac

stanimirsp5, 2023-09-06:
Has anyone succeded to run avr_debug on Mac? 
I'm using VS Code for Mac with Arduino extension and now I want to be able to debug my program but there are a lot of obstacles in configuring the launch.json file.

- First for 'miDebuggerPath' the documentation says to set the path to avr-gdb. But there isn't `avr-gdb`. I searched for it in the Arduino IDE 2.0 directory but such a file doesn't exist. So I downloaded `AVR 8-Bit Toolchain (OSX)` from [microchip](https://www.microchip.com/en-us/tools-resources/develop/microchip-studio/gcc-compilers#) website. When I run it constantly throws that error.

[image]

- My next problem is with `miDebuggerServerAddress`. What value to set there? My Arduino runs on `/dev/tty.usbmodem142401` serial port. When I set that value the vs code enters an endless loop.

- Also I can't turn off compiler optimizations because such path `hardware/arduino/avr`  in Arduino IDE 2.0 directory on Mac doesn't exist

So my biggest problem is with paths especially because documentation is strictly for Windows and it's very hard and time-consuming to search workarounds for Mac. So if anyone has succeeded in running the avr_debug on Mac, please tell me:

- where in Arduino IDE  to find 'avr-gdb'?
- what value for 'miDebuggerServerAddress' to set?
- where in Arduino IDE to place 'platform.local.txt'?

This is how my Arduino.ino file looks like 
`
#include <app_api.h>
#include <avr_debugger.h>
#include <avr8-stub.h>

void setup()
{
    debug_init();
}

void loop()
{
}
`
and this is how my launch.json file looks like
`
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387 "version": "0.2.0",
    "configurations": [
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/Arduino.ino.elf",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": true,
            "MIMode": "gdb",
            "miDebuggerPath": "/Users/stanimirpetrov/Downloads/avr8-gnu-toolchain-darwin_x86_64/bin/avr-gdb",
            "miDebuggerServerAddress": "\\\\.\\COM7",
            "setupCommands": [
                {
                    "description": "Remote debug enable",
                    "text": "-gdb-set debug remote 1",
                    "ignoreFailures": false
                }
            ]
        }
    ]
}
`

Thanks for your time.

felias-fogg, 2023-09-06:
Hi,

one way to get a decent avr-gdb on a Mac is using homebrew! Install the homebrew system and then:

brew tap osx-cross/avr && brew install avr-gdb

You can copy the binary which you find following the link at /usr/local/bin/avr-gdb to the place where you need it, or you simply set the path to that link.

The hardware folder on a Mac can either be found in the Application itself (you need to open the corresponding folder) or it is below ~/Library/Arduino15 (or both).

stanimirsp5, 2023-09-09:
Hi, @felias-fogg, thanks for your answer it was very helpful. 
I overcame my issues with the launch.json and I think everything is fine with the JSON file. This is the updated version [link](https://snappify.com/view/f397d419-06ee-4efb-ab99-e024e0c0571e)

Now my problem is that when I upload my code to Arduino and run the debugger it enters in loop and vs code begins constantly to load something. 
I thought the problem may be with `Baud rate` but I know that the debugger uses 115200 and the default baud rate for Arduino extention is 115200 too.

felias-fogg, 2023-09-10:
Hi @stanimirsp5,

I did not try it out yet - I still have to install VSC and the Arduino extension. However, the default baud rate of avr-gdb is 9600! So you need to tell avr-gdb to switch to 115200. Either use `set serial baud 115200 ` after avr-gdb started and before you establish a connection with `target remote ...` or call avr-gdb with the argument `-b 115200`.