jdolinay/avr_debug #54: Custom Atmega1284p Board Bootloader Issue

fernando-excelsense, 2023-09-08:
Hi, I am trying to use your debugger in a simple code using a custom board with and Atmega1284p, but I am not able to get it to work. I tried getting the optiboot bootloader and changing the Make file to set the following information.

What I changed:
- The clock frequency from 16000000 to 8000000
- The baud rate to 38400

```
# Atmega1284p (8MHz full-swing xtal) @ 38400 baudrate
atmega1284_38: TARGET = atmega1284p
atmega1284_38: MCU_TARGET = atmega1284p
atmega1284_38: CFLAGS += '-DLED_START_FLASHES=3' '-DBAUD_RATE=38400' '-DBIGBOOT'
atmega1284_38: AVR_FREQ = 8000000L
atmega1284_38: LDSECTIONS  = -Wl,--section-start=.text=0x1fc00
atmega1284_38: $(PROGRAM)_atmega1284p_38.hex
atmega1284_38: $(PROGRAM)_atmega1284p_38.lst

atmega1284_38_isp: atmega1284_38
atmega1284_38_isp: TARGET = atmega1284p
atmega1284_38_isp: MCU_TARGET = atmega1284p
# 1024 byte boot
atmega1284_38_isp: HFUSE = DE
# Full swing xtal (8MHz) 14CK+65ms
atmega1284_38_isp: LFUSE = F7
# 2.7V brownout
atmega1284_38_isp: EFUSE = FD
atmega1284_38_isp: isp
```

After compiling and loading this new bootloader to my device, I am able to program it, but the debugger is not working.

`main.cpp`
```
#include <Arduino.h>
#include "avr8-stub.h"
#include "app_api.h"

void setup() {
  debug_init();  
  breakpoint();
  while(1);
}

void loop() {
}
```

`platformio.ini`
```
[env:custom]
platform = atmelavr
board = custom
framework = arduino

upload_port = COM13
monitor_port = COM13
monitor_speed = 38400

debug_tool = avr-stub
debug_port = \\.\COM13
debug_init_break = setup

lib_deps =
    jdolinay/avr-debugger @ ~1.4
```

`custom.json`
```
{
  "build": {
    "core": "arduino",
    "f_cpu": "8000000L",
    "mcu": "atmega1284p",
    "variant": "custom"
  },
  "debug": {
    "simavr_target": "atmega1284p",
    "avr-stub": {
      "speed": 115200
    }
  },
  "frameworks": [
    "arduino"
  ],
  "upload": {
    "maximum_ram_size": 16384,
    "maximum_size": 130048,
    "protocol": "arduino",
    "require_upload_port": true,
    "speed": 38400
  },
  "name": "custom",
  "url": "",
  "vendor": ""
}

```

The error I am getting...
[image]

I also tried following the bootloader flash instructions in **avr_debug/bootloader/optiboot/readme_platformio.txt**, but I cannot make it work. 

Could you please let me know what I might be doing wrong and how to fix this issue? 

Thanks!

jdolinay, 2023-09-30:
Hi, sorry for late reply. 
The error message does not say much, the gdb on your computer is not able to communicate with the stub in your board. There may be many reasons.

First, I'd like to say you should be able to use the debugger without changing the bootloader. New bootloader is only needed for the "flash breakpoints" configuration. Even if you want to use this configuration it will be easier to first get it running with the default "RAM breakpoints", and this is completely independent of the bootloader. 

The baud rate for the bootloader is not related to the baud rate for the debugger. For the debugger you can select baudrate using the AVR8_USER_BAUDRATE compile option. Default is 115200 but I suspect if you use 8 MHz clock for the CPU, then the baudrate is 57600. 

You could try to set the baudrate in you platform.ini file using the debug_build_flags. Something like this:
 
debug_build_flags =  
    -Og 
    -g2  
    -DDEBUG  
    -DAVR8_USER_BAUDRATE=57600

Another problem may be how to tell gdb to use this baudrate. Probably this is in this section in custom.json:
"avr-stub": {
      "speed": 115200
    }
where you can change it to 57600.

fernando-excelsense, 2023-10-04:
Hi, thank you for the help, it is working now!